teejusb/fsr #5: Arduino Keyboard library not supported for ATmega328P, ATmega1280 or ATmega2560

Natologic, 2021-02-19:
Thanks for providing this excellent code to the community.

Unless I'm missing something, the .ino won't compile for the ATmega328P, ATmega1280 or ATmega2560 as the Arduino "Keyboard" library isn't supported on chips that don't have native USB interfaces. The ATmega32U4 has a USB interface built on the chip itself and compiles fine.

teejusb, 2021-02-19:
That is correct. Only those boards that make can make use Arduino's keyboard library will be able to use this code. The other boards won't have a way to send keystrokes/joystick inputs that StepMania can work with. According to the docs the 32u4 and SAMD based micro boards work with the Keyboard library. https://www.arduino.cc/reference/en/language/functions/usb/keyboard/

Natologic, 2021-02-19:
OK, my understanding is correct. In that case is there a reason why
https://github.com/teejusb/fsr/blob/e82cc8fa2890cd0103363233451be30d2fdfb2b5/fsr.ino#L139-L140
checks for ATmega328P, ATmega1280 or ATmega2560?

teejusb, 2021-02-19:
That was technically to skip the averaging portion of the code since it seems to not work on the 8-bit boards, so I had just grabbed a list of the 8-bit boards. You're right in that I assumed that all Arduino boards would have Keyboard functionality which I now see was an incorrect assumption.

Natologic, 2021-02-19:
Thanks for the clarification, in that case, not compiling for those boards is expected behavior. Consequently this isn’t a true issue and can be closed. 

Thanks again for providing this to the community!

closed 2021-02-19 by Natologic