teejusb/fsr #8: is there a way you can add a 5th input for pump it up?

johnboyjr, 2021-03-21:
is there a way you can add a 5th input for pump it up?

teejusb, 2021-04-10:
It's possible! You have to add a new sensor state in the SensorStates array in the fsr.ino code. And then for the webui, you'll have to change the following lines to account for the additional sensor.

I do have a plan to streamline all of this so in the future you would only need to change the SensorStates array and the rest should just work. But for now, this is what you have to do.

https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L67
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L77
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L123
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L175
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L208
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L217
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L225 (not as important)
https://github.com/teejusb/fsr/blob/master/webui/server/server.py#L250

https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L29
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L34
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L255-L258
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L267
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L268
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L326
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L347
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L363
https://github.com/teejusb/fsr/blob/master/webui/src/App.js#L395-L409

DJKero, 2021-08-02:
Commit: https://github.com/DJKero/fsr/commit/d7477b667e511e56de004e07fd941abf4ccefbac

I gave it a shot but I won't be able to test it for a while since I'm waiting for the stuff to arrive.

teejusb, 2022-02-03:
I think this is solved with #22 so will close.

closed 2022-02-03 by teejusb