teejusb/fsr #14: Info file .EXE for Windows

MircoLeo, 2021-05-23:
hi, I wanted to ask if you will create an installation file for windows for the management of the FSR thresholds, in order to make it easier to manage the threshold change of each sensor, and if you increase the firmware for arduino, you will be able to configure the 16 sensors configuration that you can then possibly manage the calibration from a single program that can be installed on windows

teejusb, 2021-05-23:
I don't quite follow. There's a WebUI that lets you manage the thresholds already.

MircoLeo, 2021-05-23:
I understand, it was to know if it was in progress and to make it easier even for those not familiar with programming, the threshold management individually through a dedicated program ^ _ ^

DJKero, 2021-08-04:
> 
> 
> I don't quite follow. There's a WebUI that lets you manage the thresholds already.

I think he may be asking for a Windows Installer that:

1. Stores the repo in a configurable location.
2. Installs Python, Node and Yarn. (Steps 1 & 2 of the UI setup)
3. Asks you which Microcontroller you're using and amount of panels to modify Server.py.
4. Executes a batch script that automates steps 4 through 9 of the UI setup. (using the location from point 1)

So you only need to do the firmware setup. Is that what you're asking for @MircoLeo?

teejusb, 2021-08-04:
It's confusing because he states `in order to make it easier to manage the threshold change of each sensor` which is what the WebUI does.

A (cross platform) installer for setup is something I do want but I wasn't sure if that's what he was asking. Also there is an included batch script in the readme if someone wants an easy way to start up the WebUI.

DJKero, 2021-08-04:
Yeah I know, the wording is a bit confusing, I'm not sure if he thinks the WebUI setup is too cumbersome or if he's not aware of it, I lean towards the former.
And I think he's talking about the addition of sensors/panels too.

MircoLeo, 2021-08-05:
exactly, my request was to install the program in windows instead of webui since I don't know how to use Python, Node and Yarn to manage the thresholds, I only know how to install the arduino file and that's it

> > Non seguo molto. C'è una WebUI che ti consente già di gestire le soglie.
> 
> Penso che potrebbe chiedere un programma di installazione di Windows che:
> 
> 1. Memorizza il repository in una posizione configurabile.
> 2. Installa Python, Node e Yarn. (Fasi 1 e 2 della configurazione dell'interfaccia utente)
> 3. Ti chiede quale Microcontrollore stai utilizzando e la quantità di pannelli da modificare Server.py.
> 4. Esegue uno script batch che automatizza i passaggi da 4 a 9 della configurazione dell'interfaccia utente. (usando la posizione dal punto 1)
> 
> Quindi devi solo eseguire la configurazione del firmware ed eseguire il programma di installazione. È questo che stai chiedendo per @MircoLeo ?

> È confuso perché afferma `in order to make it easier to manage the threshold change of each sensor`che è ciò che fa la WebUI.
> 
> Un programma di installazione (multipiattaforma) per l'installazione è qualcosa che voglio, ma non ero sicuro che fosse quello che chiedeva. Inoltre c'è uno script batch incluso nel readme se qualcuno vuole un modo semplice per avviare la WebUI.

DJKero, 2021-08-05:
> 
> 
> exactly, my request was to install the program in windows instead of webui since I don't know how to use Python, Node and Yarn to manage the thresholds, I only know how to install the arduino file and that's it

I see, yeah the setup process of the whole thing is pretty involved. An installer for the WebUI so it's easier to setup is a good idea.
@teejusb do you know of a cross-platform installer solution to look into?

MircoLeo, 2021-08-06:
> > appunto, la mia richiesta era di installare il programma in windows invece che webui visto che non so usare Python, Node e Yarn per gestire le soglie, so solo installare il file arduino e basta
> 
> Vedo, sì, il processo di installazione dell'intera faccenda è piuttosto complicato. Un programma di installazione per l'interfaccia utente Web in modo che sia più semplice da configurare è una buona idea.
> @teejusb conosci una soluzione di installazione multipiattaforma da esaminare?

thi is ?
https://github.com/electromuis/analog-dance-pad/releases

teejusb, 2021-08-07:
I do have an installer in progress so I'll get it out soon enough.

MircoLeo, 2021-08-09:
> Ho un programma di installazione in corso, quindi lo tirerò fuori abbastanza presto.

good
as soon as ready I will install your firmware together with the executable program, so if it is plug and play for every pc I use it would be great

teejusb, 2025-05-09:
I don't think I'll finish the installer anytime soon, but as a stop gap you can make use of this other webui which removes the need to setup your own WebUI

https://emkooz.github.io/webfsr/
https://github.com/emkooz/webfsr

closed 2025-05-09 by teejusb