teejusb/fsr #28: Installing the latest version will throw build issues when using pip

inyongthegr8, 2021-12-08:
Tried installing the latest version of Python (3.10.1), experiencing build wheel failures for multidict and yarl.

For now, the stable way to install Python is by installing [Python 3.8.6](https://stackoverflow.com/a/64452019).

teejusb, 2022-02-03:
I commited https://github.com/teejusb/fsr/commit/52b3616afe0c2c255a251c406327127e2fa26d38. Do you think you can check if it works on 3.10 for you now?

inyongthegr8, 2022-02-03:
> I commited [52b3616](https://github.com/teejusb/fsr/commit/52b3616afe0c2c255a251c406327127e2fa26d38). Do you think you can check if it works on 3.10 for you now?

Hello, @teejusb!

Thank you for working on this, apparently, I tried it on my machine and it didn't work.

Here's the whole log:
`G:\Documents\jhomz\fsr-master\webui\server>py -m venv venv

G:\Documents\jhomz\fsr-master\webui\server>venv\Scripts\activate

(venv) G:\Documents\jhomz\fsr-master\webui\server>pip install -r requirements.txt
Collecting aiohttp==3.7.4.post0
  Downloading aiohttp-3.7.4.post0.tar.gz (1.1 MB)
     |████████████████████████████████| 1.1 MB 819 kB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting async-timeout==3.0.1
  Downloading async_timeout-3.0.1-py3-none-any.whl (8.2 kB)
Collecting attrs==21.2.0
  Downloading attrs-21.2.0-py2.py3-none-any.whl (53 kB)
     |████████████████████████████████| 53 kB 943 kB/s
Collecting chardet==4.0.0
  Downloading chardet-4.0.0-py2.py3-none-any.whl (178 kB)
     |████████████████████████████████| 178 kB ...
Collecting idna==3.2
  Downloading idna-3.2-py3-none-any.whl (59 kB)
     |████████████████████████████████| 59 kB ...
Collecting multidict==5.1.0
  Downloading multidict-5.1.0.tar.gz (53 kB)
     |████████████████████████████████| 53 kB 931 kB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting pyserial==3.4
  Downloading pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |████████████████████████████████| 193 kB ...
Collecting typing-extensions==3.10.0.0
  Downloading typing_extensions-3.10.0.0-py3-none-any.whl (26 kB)
Collecting yarl==1.7.2
  Downloading yarl-1.7.2-cp310-cp310-win_amd64.whl (122 kB)
     |████████████████████████████████| 122 kB ...
Building wheels for collected packages: aiohttp, multidict
  Building wheel for aiohttp (PEP 517) ... done
  Created wheel for aiohttp: filename=aiohttp-3.7.4.post0-py3-none-any.whl size=451945 sha256=5efc1da13c77172b80e063dea7c539915172dec90430b65be05255fd94a6f6d9
  Stored in directory: c:\users\inyon\appdata\local\pip\cache\wheels\fc\84\e0\52113c57eb9b09b6b187a0f369eaad6fc7fc64bb7247c83b89
  Building wheel for multidict (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: 'G:\Documents\jhomz\fsr-master\webui\server\venv\Scripts\python.exe' 'G:\Documents\jhomz\fsr-master\webui\server\venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\inyon\AppData\Local\Temp\tmpnvsx14x8'
       cwd: C:\Users\inyon\AppData\Local\Temp\pip-install-h8g68mt0\multidict_4ad3fb6de61a4d1fa4c735d0299287e5
  Complete output (81 lines):
  *********************
  * Accelerated build *
  *********************
  running bdist_wheel
  running build
  running build_py
  running egg_info
  writing multidict.egg-info\PKG-INFO
  writing dependency_links to multidict.egg-info\dependency_links.txt
  writing top-level names to multidict.egg-info\top_level.txt
  reading manifest file 'multidict.egg-info\SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files matching '*.pyc' found anywhere in distribution
  warning: no previously-included files found matching 'multidict\_multidict.html'
  warning: no previously-included files found matching 'multidict\*.so'
  warning: no previously-included files found matching 'multidict\*.pyd'
  warning: no previously-included files found matching 'multidict\*.pyd'
  no previously-included directories found matching 'docs\_build'
[... 45 lines cut ...]
      self.build_extensions()
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 448, in build_extensions
      self._build_extensions_serial()
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 473, in _build_extensions_serial
      self.build_extension(ext)
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\command\build_ext.py", line 202, in build_extension
      _build_ext.build_extension(self, ext)
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 528, in build_extension
      objects = self.compiler.compile(sources,
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 327, in compile
      self.initialize()
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 224, in initialize
      vc_env = _get_vc_env(plat_spec)
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\msvc.py", line 316, in msvc14_get_vc_env
      return _msvc14_get_vc_env(plat_spec)
    File "C:\Users\inyon\AppData\Local\Temp\pip-build-env-t99kwro1\overlay\Lib\site-packages\setuptools\msvc.py", line 270, in _msvc14_get_vc_env
      raise distutils.errors.DistutilsPlatformError(
  setuptools._distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for multidict
Successfully built aiohttp
Failed to build multidict
ERROR: Could not build wheels for multidict which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 21.2.4; however, version 22.0.3 is available.
You should consider upgrading via the 'G:\Documents\jhomz\fsr-master\webui\server\venv\Scripts\python.exe -m pip install --upgrade pip' command.`

teejusb, 2022-02-05:
I pushed an update to multidict as well which should work with 3.10. Think you could try that as well?