teejusb/fsr #102: Update README.md to correct RP2040 guide and also to correct Windows guide

sukibaby, 2025-04-21:
There are two issues...

1.  There is no reference to the needed RP2040 Arduino Core: https://github.com/earlephilhower/arduino-pico
2.  Microsoft has ceased distributing Visual C++ Build Tools standalones. Users must now install these via the Visual Studio Installer **before running pin install -r requirements.txt** or installing `aiohttp` will not succeed.
              - Windows SDK (11.22000 is fine)
              - latest MSVC Build Tools  for x64/x86

teejusb, 2025-04-22:
regarding 2, couldn't you just grab Python from the Microsoft store or the python site directly https://www.python.org/downloads/windows/? That's all you really need to do to run `pip install`. I personally did the former and seems to work fine. Never really even crossed my mind to go through visual studio since it didn't seem necessary.

[image]

sukibaby, 2025-04-22:
I'm confused because I obviously have python installed or else how would I be in the venv to run `pip install -r requirements.txt`?  Python is not the issue here.

The problem is when aiohttp gets installed by pip. If you don't have Visual C++ Build Tools it can't install aiohttp and it fails out. You used to be able to follow a link from the error message, get the standalone Visual C++ Build Tools, and then re-run `pip install -r requirements.txt`.

You can't download a standalone Visual C++ Build Tools anymore. If you go to any link to download that from the Microsoft site, it redirects you to Visual Studio Installer. So point number 2 in my original post is simply noting how one installs the correct Visual C++ Build Tools via Visual Studio Installer.

sukibaby, 2025-04-22:
For further context: the reason this issue exists is because a friend asked me to help them get the FSR webui set up on their Windows 11 laptop because they couldn't figure it out. I had them bring the laptop over and went thru the setup instructions myself. The pip install command was failing early because it wasn't building aiohttp due to the missing MSVC stuff.

teejusb, 2025-04-22:
Aahh I see what you mean now. I probably already have the distributable installed on my system which is why I haven't run into it. I assume just linking: https://visualstudio.microsoft.com/visual-cpp-build-tools/ is probably fine? Additionally if you have screenshots I can link those in the readme as well.

sukibaby, 2025-04-22:
That's perfectly fine. Yes I also already would have had the needed MSVC stuff on my PC I use for ITGm dev so I never would have noticed it either.  I don't have screenshots atm but I can make one.

sukibaby, 2025-04-22:
[image]

[image]

feel free to crop further if desired

either Windows 10 SDK or Windows 11 SDK is acceptable for this purpose, user can use whatever matches their OS

sukibaby, 2025-05-18:
There is actually a significantly easier way to get the necessary build tools to avoid all this headache with the vc2019 build tools. In the Node.js setup, there's an option which is disabled by default to install necessary stuff to build the C++. 

[image]

If user checks that checkbox, it handles all the dependencies automatically.